Qudor-Engineer/DMD32 #1: Crash when used alongside BLE libraries

BodoMinea, 2021-01-07:
Hi,

I just wanted to ask if you have tried using this library alongside some Bluetooth (BLE) functionality. The test sketch examples for fine, but after adding the BLE dependencies
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
```
and some code that listens for a connection, the ESP32 just boot-loops with an error.

freertos/queue.c:1446 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008ddf7 on core 1
Rebooting...

Is there some forceful limitation of clock speed in DMD32 or something that could make the Bluetooth peripheral fail to initialize? The library works fine otherwise, albeit with some quirks (I found that if I try display something on the dot matrix display from Setup or from Loop but without a delay after, the ESP will crash and reboot, just like when adding Bluetooth to an otherwise working code).

Thanks!

Qudor-Engineer, 2021-01-12:
Unfortunately I didn't try the library with the built-in ble

Qudor-Engineer, 2021-03-12:
The problem with the hardware timer, just change the hardware timer from 0 to 1 or 2

BodoMinea, 2021-03-23:
I will try that when I have the panels on hand and come back with feedback. Thanks.

BodoMinea, 2021-03-31:
I am not sure if this was what you are suggesting but I changed `timer = timerBegin(0, cpuClock, true);` to `timer = timerBegin(1, cpuClock, true);` in my sketch and saw some improvement - now the crash is not on startup and I can display some stuff with dmd while still using bluetooth. However, after some time the code crashes with the same message.

I will also try hardware timer 2 some time later.

Do you know if there is some other hardcoded mention or use of the 0 timer?

microElabDevelop, 2022-06-20:
@BodoMinea hi u find any solution about crashing. bcz i am facing same issue . i m using dmd32 with espNow . for wireless display.