Qudor-Engineer/DMD32 #8: Matrix of 3 x 5 displays

leonhutton, 2021-02-23:
Hi,

Thanks for creating this library.  I am trying to adjust the library to work with DISPLAYS_ACROSS 3 and DISPLAYS_DOWN 5.  

I have done some experimentation but I my main issue is understanding the following function:

`DMD::DMD(byte panelsWide, byte panelsHigh)
{
    uint16_t ui;
    DisplaysWide=panelsWide;
    DisplaysHigh=panelsHigh;
    DisplaysTotal=DisplaysWide*DisplaysHigh;
    row1 = DisplaysTotal<<4;
    row2 = DisplaysTotal<<5;
    row3 = ((DisplaysTotal<<2)*3)<<2;
    bDMDScreenRAM = (byte *) malloc(DisplaysTotal*DMD_RAM_SIZE_BYTES);`

Could this be adjusted to accommodate another row?

Thanks for your help!

leonhutton, 2021-02-23:
One other thing to note: I have also tried to connect my displays in a 4 x 4 array, but I get a flashing display.  Any combination up to 4 rows works, but adding the 4th row makes the display flash.

Qudor-Engineer, 2021-03-12:
Hi,
Please refer to the original DMD library (https://github.com/freetronics/DMD).

closed 2021-11-18 by Qudor-Engineer