Qudor-Engineer/DMD32 #13: Not work espressinf32@5.0.0

MaxSKash, 2022-07-09:
Compiled, not working.
Working espressinf32@3.5.0

manumdk, 2023-02-26:
Hi,
I have the same problem, did you solve it ?

manumdk, 2023-02-26:
This fork is OK with espressif32@5.1.0
[https://github.com/MiyukiDark/DMD32](url)

fernandoyano, 2023-05-19:
Hi,
the problem is only on init SPI, because MISO is use as OUTPUT on library, I use in espressif32@6.2.0

arrowmoon2000, 2025-10-26:
thanks, https://github.com/Qudor-Engineer/DMD32/issues/13#issuecomment-1445430783
very helpfull

P10 led panels works very well with changed below pin modes/codes and I tested bye latest (26.10.2025) 3.3.2 ESP32 library

additionally , I added led panels brightness feaature , all panel leds (changable brightness) work very well, below link 
https://github.com/Qudor-Engineer/DMD32/issues/29#issuecomment-3448350171

[image]

Update pin modes in the DMD32.cpp

in DMD::DMD(byte panelsWide, byte panelsHigh) 

    pinMode(PIN_DMD_A, OUTPUT);
    pinMode(PIN_DMD_B, OUTPUT);
    pinMode(PIN_DMD_SCLK, OUTPUT);
    pinMode(PIN_DMD_nOE, OUTPUT);

    digitalWrite(PIN_DMD_A, LOW);
    digitalWrite(PIN_DMD_B, LOW);
    digitalWrite(PIN_DMD_SCLK, LOW);
    digitalWrite(PIN_DMD_nOE, LOW);