Qudor-Engineer/DMD32 #21: P10 not working with ESP

Mahadkz, 2023-07-05:
I am trying to work with HUB12 matrices monochrome with P10 but they just do not work no matter what code I try the displays just respond the same they turn on all the LEDs. For example I tried this `#include <DMD32.h> 
#include "fonts/SystemFont5x7.h"
#include "fonts/Arial_Black_16_ISO_8859_1.h"

//Fire up the DMD library as dmd
#define DISPLAYS_ACROSS 2
#define DISPLAYS_DOWN 1
DMD dmd(DISPLAYS_ACROSS, DISPLAYS_DOWN);

   //Timer setup
  //create a hardware timer  of ESP32
  hw_timer_t * timer = NULL;

void IRAM_ATTR triggerScan()
{
  dmd.scanDisplayBySPI();
}

void setup(void)
{

// return the clock speed of the CPU
  uint8_t cpuClock = ESP.getCpuFreqMHz();
  
  // Use 1st timer of 4 
  // devide cpu clock speed on its speed value by MHz to get 1us for each signal  of the timer
  timer = timerBegin(0, cpuClock, true);
  // Attach triggerScan function to our timer 
  timerAttachInterrupt(timer, &triggerScan, true);
  // Set alarm to call triggerScan function  
  // Repeat the alarm (third parameter) 
  timerAlarmWrite(timer, 300, true);

  // Start an alarm 
  timerAlarmEnable(timer);

   //clear/init the DMD pixels held in RAM
   dmd.clearScreen( true );   //true is normal (all pixels off), false is negative (all pixels on)
  Serial.begin(115200);
}

void loop(void)
{
   dmd.clearScreen( true );
   dmd.selectFont(Arial_Black_16_ISO_8859_1);
  
   const char *MSG = "This Works";
   dmd.drawMarquee(MSG,strlen(MSG),(32*DISPLAYS_ACROSS)-1,0);
   long start=millis();
   long timer=start;
   while(1){
     if ((timer+30) < millis()) {
       dmd.stepMarquee(-1,0);
       timer=millis();
     }
   }
}` same result then I tried this `#include "DMD32.h"

// Create an instance of DMD
DMD dmd(2, 1); // Assuming 2 panels wide and 1 panel high

void setup() {
  // Initialize DMD
  dmd.clearScreen(true);
}

void loop() {
  // Write a pixel at coordinates (x, y)
  int x = 10;
  int y = 5;
  byte graphicsMode = GRAPHICS_NORMAL;
  byte pixelValue = 1; // Set to 0 for clearing the pixel, or 1 for setting the pixel
  dmd.writePixel(x, y, graphicsMode, pixelValue);

  // Delay for some time
  delay(1000);
}` I even tried examples that come with Libraries but It doesn't work
the connections are according to the DMD32 Library
'#define PIN_DMD_nOE		22 		// D22 active low Output Enable, setting this low lights all the LEDs in the selected rows. Can pwm it at very high frequency for brightness control.
#define PIN_DMD_A		19		// D19
#define PIN_DMD_B		21		// D21
#define PIN_DMD_CLK		18		// D18_SCK  is SPI Clock if SPI is used
#define PIN_DMD_SCLK		2		// D02
#define PIN_DMD_R_DATA    23	// D23_MOSI is SPI Master Out if SPI is used'
I tried changing panels both are okay used one at a time and both as well I am using ESP32 DEV Kit. 
Any help will be appreciated

RARadchenko, 2023-11-05:
have same problem did you solve it?

FrenzyExists, 2024-05-08:
Too late of a reply, but I think the issue has to do with the chip the HUB12 uses. I never used that display before, I used the p10, but still, check the mcus that manage the matrix, if its not the same as the p10 then that could be the reason it never worked