Qudor-Engineer/DMD32 #22: ESP32 DEVKITV1 doesn't work with this library

galagen8, 2023-07-07:
The library doesn't work. I tried with ESP32 DEVKITV1. I used correct pinout and example sketches, checked and re-checked everything several times, but no any of LEDs were lights up. I tried same DMD panel with Arduino Nano and ESP8622 to make sure the panel isn't broken — and it's working. Did you change something in the library?

dirtyfunky, 2023-08-04:
I also try to use DEV kit with this library but my panels are showing 8 horizontal lines 4 bright, 4 slightly dimmed. but program is not working correctly.
I was able to run this panels with Arduino UNO with DMD library, but wanted to use esp32 because of speed.

platro, 2025-06-23:
Успешно работает на версии 3.0.4. В фунции void DMD::scanDisplayBySPI() нужно убрать условие if( digitalRead( PIN_OTHER_SPI_nCS ) == HIGH )

thisaru500, 2025-09-17:
It is esp-32 version issue you need to change your version to V1.0.0 
Because issue is this library timer is not valid for new versions

platro, 2025-09-17:
А так же, начиная со строки 56 файла DMD32.cpp нужно изменить код на следующий:
    pinMode(PIN_DMD_A, OUTPUT);	//
    pinMode(PIN_DMD_B, OUTPUT);	//
    //pinMode(PIN_DMD_CLK, OUTPUT);	//
    pinMode(PIN_DMD_SCLK, OUTPUT);	//
    //pinMode(PIN_DMD_R_DATA, OUTPUT);	//
    pinMode(PIN_DMD_nOE, OUTPUT);	//
    

    digitalWrite(PIN_DMD_A, LOW);	// 
    digitalWrite(PIN_DMD_B, LOW);	// 
    //digitalWrite(PIN_DMD_CLK, LOW);	// 
    digitalWrite(PIN_DMD_SCLK, LOW);	// 
    //digitalWrite(PIN_DMD_R_DATA, HIGH);	// 
    digitalWrite(PIN_DMD_nOE, LOW);	//
Потому что пины CLK и R_DATA не являются GPIO пинами, это SPI пины.