Qudor-Engineer/DMD32 #24: p10 panel with ESP32 DEV

thilankapun, 2023-09-24:
Hi 
im trying this lib to work with ESP32, but nothing in the display , checked the clock signal but there is no clk signal output in pin 18 no response with LED panel.

RARadchenko, 2023-11-05:
Have same problem. Did you solved it?

thilankapun, 2023-11-05:
hi
no , it is not solved yet , some of led are blinking its a problem of library and tried to figured out but its not successful.

RARadchenko, 2023-11-05:
i`m solved it. If you have telegram or somesthing else just write me, i show you my wiring and main problem (tg: @Atomic470)
[image]

thilankapun, 2023-11-05:
great , can you share the details with email pls, i dont use telegram.

vivek9676, 2023-11-06:
Hi, I am facing the same issue. Can you help me?
telegram on: +918962307685

MMB923, 2023-12-27:
Hello
I have a similar problem using the P10 module.
I sent you an email and hope to receive the answer as soon as you can.
thanks
mohammadmahdibayat88@gmail.com

RARadchenko, 2023-12-28:
> Hello I have a similar problem using the P10 module. I sent you an email and hope to receive the answer as soon as you can. thanks [mohammadmahdibayat88@gmail.com](mailto:mohammadmahdibayat88@gmail.com)

https://github.com/Qudor-Engineer/DMD32/issues/25

MMB923, 2024-01-01:
> [](https://github.com/Qudor-Engineer/DMD32/issues/25)

I'm not using Arduino or ESP module. I use STM32 processor like STM32F4
Any recommendation?

tanish0305, 2024-10-09:
Use the ESP32 Board version 2.0.2.
It will work.