Qudor-Engineer/DMD32 #25: Solution in case if your matrix doesn`t work

RARadchenko, 2023-11-06:
I`m using esp32 devkit v1 and P10 WO (single color) andmain solution is downgrading esp kernel to 2.0.2 version. On 2.0.14 ver CLK LAT and DR pin was 0V.
[image]
in board manager choose more older version for esp32

here is my wiring
[image]
brown wire is DR (or R on another version of p10 matrix) connected to D23 on esp32
blue wire is LAT (or LCLK) connected to D2 
green wire is CLK connected to D18
yellow wire is B connected to D21
red wire is A connected to D19
purple wire is OE (or enable) connected to D22
black wire is gnd connected to any gnd esp`s pin

[image]
Paint wiring diagramm

here is normal voltage
[image]
OE pin
[image]
A pin
[image]
B pin
[image]
CLK pin
[image]
LAT pin
[image]
DR pin

Adityavns, 2024-03-18:
Any ideas on how to make it work on latest versions and using platformio?

MMB923, 2024-04-21:
I have a library for Arduino Uno which I've tested for several times. but I need a good one for STM32 controllers using CubeIDE or Keil.
Thanks

tientruong611, 2024-06-29:
u're a KING sir. I spend 1 week to finding problem , now i was found that. thanks u very much. I have a question, may i build the library on my own and don't using the library DMD

Suresh6060, 2024-10-15:
It works with 1.0.5 version . I wonder how lowering the version shall affect other libraries.

khattarrohit, 2025-03-06:
By ESP32 Can we connect 3 panel in a single row with it .

Suresh6060, 2025-03-06:
khattarrohit, 

yes you can easily. I have connected 4 panels in a single row

khattarrohit, 2025-03-06:
> khattarrohit,
> 
> yes you can easily. I have connected 4 panels in a single row

I will work on it. please be in touch with us for any other issue like it's program.
Thanks for your reply.

khattarrohit, 2025-05-17:
Sir,
    I am writing this comment with full hope and expectation from you. 
  I want to display AQI data from Open source website API by using ESP32 on 3 P10 Modules in series.
       Please help me to make a program of it in arduino ide.
                            I am waiting of your reply.
                                           Thankyou

nminhduc-sonam, 2025-05-22:
> Sir, I am writing this comment with full hope and expectation from you. I want to display AQI data from Open source website API by using ESP32 on 3 P10 Modules in series. Please help me to make a program of it in arduino ide. I am waiting of your reply. Thankyou

where do you live, bro? I can help you, I have display up to 20 led panels p10.

platro, 2025-06-23:
Успешно работает на версии 3.0.4. В фунции void DMD::scanDisplayBySPI() нужно убрать условие if( digitalRead( PIN_OTHER_SPI_nCS ) == HIGH )